espressif/esp-usb-bridge #1: Idea: ESP32 as GDB remote over WiFi

ppisa, 2022-04-16:
The first thanks for the nice project. USB JTAG API is generic and provides support to almost every possible OpenOCD supported target but if the subset of OpenOCD or some other GDB remote compatible server is integrated directly into ESP32, ideally NuttX in general, then ESP32 or even other micros can be used to debug system over WiFi or Ethernet. It has big advantage of the galvanic isolation for real world industrial targets and may it be can be even faster then USB...

igrr, 2022-04-16:
@ppisa thank you for the suggestion. Technically it is possible to cross-compile OpenOCD project (with not a lot of changes) and run it on the esp32. Given that NuttX provided comparable number of POSIX APIs, I think it should be possible to integrate OpenOCD into a NuttX app as well.

The issue is that OpenOCD is GPL v2 licensed, while this project (and the rest of IDF) is Apache 2.0. The two licenses [are incompatible](https://www.apache.org/licenses/GPL-compatibility.html), unfortunately.
I think NuttX is also Apache licensed now, so it's the same situation there.

Some people have suggested that the SDK (IDF or NuttX) can in this case be exempt under "system library exception" clause of GPLv2, but this is subject to interpretation.

One alternative I know of is https://github.com/walmis/blackmagic-espidf.

ppisa, 2022-04-16:
Thanks for the comment. In a NuttX case, OpenOCD lite can be compiled as standalone application so the NuttX will serve purpose of run-time environment which would not be a problem from GPL point of view even that APIs are referenced by functions calls not syscalls.

OpenOCD is declared to be SPDX-License-Identifier: GPL-2.0-or-later which means that combined work of Appache licensed NuttX or even IDF could be delivered under GPL 3 license... if there are no binary blobs redistributed... Which can be problem for WiFi.

Anyway it probably means more work than profit when BlackMagic provides exactly suggested functionality. It would be great to have its port to NuttX, because it can run even on broader range of the targets. But ESP32 are nice that combine WiFi directly and are small.

By the way, we consider CAN/CANopen NuttX "probe" based on ESP32C3, now mainlined TWAI driver and [OrtCAN](http://ortcan.sourceforge.net/). I hoped to find some student/novice to work on that in the frame of this year GSoC at NuttX but I have not succeed in looking for somebody interested in this. But I have not much time to advertise. Anyway I try offer that as next year theses topic at our university. 

Best wishes,

Pavel

tyeth, 2026-09-15:
Just to include that the walmis/blackmagic-espidf repo then offers the S2 compatible version in theory with better networking support here:
https://github.com/flipperdevices/blackmagic-esp32-s2