espressif/esp-usb-bridge #2: JTAG/SWD with SPI peripheral?

huming2207, 2022-04-17:
Hi there,

Firstly thanks to the great project, the USB MSC protocol looks pretty nice and clean and helped me a lot in understanding how it works. I've seen this project uses bit-banging for JTAG. But since ESP32's GPIO is [relatively slow](https://www.esp32.com/viewtopic.php?t=1595), should we consider moving to SPI-based JTAG instead?

I've also done some research earlier for my [SoulInjector SWD programmer](https://github.com/huming2207/soul-injector) project before, and I did manage to send some SWD stuff out (the reset sequence) via the SPI peripheral. However it didn't work for some unknown reasons, so I've scrapped it and stick with the bitbang solution for now. 

See here: 

- The "SWD over SPI" code: https://github.com/huming2207/soul-injector/blob/638f8e124572be082a4c027dfff0b5db3fef0a5c/main/driver/swd_driver.cpp
- Related tweet: https://twitter.com/huming2207/status/1439089050841849856

Regards,
Jackson

igrr, 2022-04-17:
Hi @huming2207, thanks for sharing the code and the suggestion! We have some WIP update which replaces gpio bitbanging with "dedicated GPIO" instructions supported on both S2 and S3. This should give performance comparable to SPI.

That said, SPI has the benefit of possibly freeing up the CPU to do other things during lengthy transactions (especially on the single core S2), so we will consider that as well!

huming2207, 2022-04-17:
@igrr yea I think SPI should be better as it somewhat frees up a bit more CPU cycles, and we can do more stuff (e.g. GDB server for this project, or speak to a production line management cloud service) with a single ESP32-S3 + WiFi/BLE.

Meanwhile, I've just found this a moment ago, someone made an ESP8266-based DAPLink, with SPI: https://github.com/windowsair/wireless-esp8266-dap/blob/master/components/DAP/source/spi_op.c

So I think it's possible to get SWD over SPI to work on ESP32. I will have a look later at mine and revise why my driver code didn't work. For JTAG, I guess it should be easier as it doesn't need to do the master/slave switching.

dobairoland, 2023-02-17:
Just for future reference, this is the mentioned commit: 1b138e491d7070a3cdac361584e009b5c45ecbde

huming2207, 2023-02-26:
Nice! I will have a try later!

Maybe the next thing I can try implementing is the SPI-based JTAG. I've tried that for SWD before but somehow the timing was wrong and the target microcontroller (STM32L031) didn't reply back to me. Perhaps I should try using SPI peripheral for JTAG again.

closed 2023-02-26 by huming2207

huming2207, 2025-02-26:
Looks like @windowsair 's wireless DAP project has updated this SPI-based SWD implementation and it supports ESP32-S3 now. See: https://github.com/windowsair/wireless-esp8266-dap/blob/master/components/DAP/source/spi_op.c 

I will see what will happen if I port this to my Soul Injector programmer. Currently it runs a dumb bitbang, ended up hitting 61KB/s programming speed on a STM32WLE5CC with my optimised flash algorithm:

[image]

It's great but I doubt it may cause timing issue if I run more complicated stuff on the host ESP32-S3. I need a hardware-based solution like that wireless DAP's SPI implementation.