espressif/esp-usb-bridge #3: unable to work normally

wuxx, 2022-04-25:
Hello, I encountered some problems during the test, which made it impossible to work normally. Have tested different hardware, initially suspect a firmware issue.

debugger: ESP32-S2 dev board 
target-board: ESP32-S2 dev board
power: power by debugger, 5V or 3V3 both tried
wire-connection: GND VCC TCK TMS TDI TDO, use dupont Line 30cm and 10cm, both tried.
os: linux (raspberry pi 4B) and windows (win10) both tried.
IO: I modify main/io.h and tried some different IO config , still fail.

any some good suggestions? I will keep testing.
I tried a CMSIS-DAP based debugger, it's no problem.
the succ log
```
pi@raspberrypi:~/oss/openocd-esp32 $ sudo ./src/openocd -s /home/pi/oss/openocd-esp32/tcl -f tcl/interface/cmsis-dap.cfg -f tcl/target/esp32s2.cfg
Open On-Chip Debugger  v0.11.0-esp32-20220411-5-g03cd2031 (2022-04-25-09:45)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter speed' not 'adapter_khz'
adapter speed: 10000 kHz

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: JTAG supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: FW Version = 0255
Info : CMSIS-DAP: Serial# = 08000001003900104b0000134e4d3434a5a5a5a597969908
Info : CMSIS-DAP: Interface Initialised (JTAG)
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 1 TDO = 1 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 10000 kHz
Info : cmsis-dap JTAG TLR_RESET
Info : cmsis-dap JTAG TLR_RESET
Info : JTAG tap: esp32s2.cpu tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32s2: Debug controller was reset.
Info : esp32s2: Core was reset.
Info : starting gdb server for esp32s2 on 3333
Info : Listening on port 3333 for gdb connections

```

some fail logs for reference.
```
pi@raspberrypi:~/oss/openocd-esp32 $ sudo ./src/openocd -s /home/pi/oss/openocd-esp32/tcl -f tcl/board/esp32s2-bridge.cfg
Open On-Chip Debugger  v0.11.0-esp32-20220411-5-g03cd2031 (2022-04-25-09:45)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1002
Info : esp_usb_jtag: capabilities descriptor set to 0x30a
Info : esp_usb_jtag: target chip id set to 2
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: serial (84F703DE6EE8)
Info : esp_usb_jtag: Device found. Base speed 750KHz, div range 1 to 1
Info : clock speed 750 kHz
Info : JTAG tap: esp32s2.cpu tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : TAP auto0.tap does not have valid IDCODE (idcode=0x0)
Info : TAP auto1.tap does not have valid IDCODE (idcode=0x0)
Info : TAP auto2.tap does not have valid IDCODE (idcode=0x0)
Info : TAP auto3.tap does not have valid IDCODE (idcode=0x0)
Info : TAP auto4.tap does not have valid IDCODE (idcode=0x0)
Info : TAP auto5.tap does not have valid IDCODE (idcode=0x0)
Info : TAP auto6.tap does not have valid IDCODE (idcode=0x0)
Info : TAP auto7.tap does not have valid IDCODE (idcode=0x0)
Info : TAP auto8.tap does not have valid IDCODE (idcode=0x0)
Info : TAP auto9.tap does not have valid IDCODE (idcode=0x0)
[... 88 lines cut ...]
F:\20220425\openocd-esp32>.\bin\openocd.exe -f  share\openocd\scripts\board\esp32s2-bridge.cfg
Open On-Chip Debugger  v0.11.0-esp32-20220411 (2022-04-11-08:48)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1002
Info : esp_usb_jtag: capabilities descriptor set to 0x30a
Info : esp_usb_jtag: target chip id set to 2
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: serial (84F703DE6EE8)
Info : esp_usb_jtag: Device found. Base speed 750KHz, div range 1 to 1
Info : clock speed 750 kHz
Error: JTAG scan chain interrogation failed: all zeroes
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: esp32s2.cpu: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Warn : target esp32s2 examination failed
Info : starting gdb server for esp32s2 on 3333
Info : Listening on port 3333 for gdb connections
shutdown command invoked
```

wuxx, 2022-04-25:
also try flash two ESP32-S2 board as debugger, still same failure.

wuxx, 2022-04-25:
and just notice that the   `adapter speed 40000` in esp_usb_bridge.cfg doesn't work, no matter how you adjust the parameters,  the tck is always about 2MHz, max freq is about 7MHz

[image]

wuxx, 2022-04-25:
hello, @erhankur here is the schematic https://github.com/wuxx/nanoESP32-S2/blob/master/schematic/nanoESP32S2-v1.2.pdf

erhankur, 2022-04-25:
Just for a reference,  I connected two ESP32-S2 boards with the below default pin setup.  

```
#define CONFIG_BRIDGE_GPIO_TDI 10  ----> TARGET  GPIO40 - TDO
#define CONFIG_BRIDGE_GPIO_TDO 9 ----> TARGET  GPIO41 - TDI
#define CONFIG_BRIDGE_GPIO_TCK 8 ----> TARGET  GPIO39 - TCK
#define CONFIG_BRIDGE_GPIO_TMS 7 ----> TARGET  GPIO42 - TMS
GND --> TARGET GND
```

And about the clock speed,  we don't support the selective clock for now. It is in our roadmap.

wuxx, 2022-04-26:
hello @erhankur ，just try your IO config and it test pass. and I check the code and datasheet, suddenly found the TDI and TDO pin define seems swapped. I think the TDI and TDO should be swapped, the TDI is output, and TDO  is input. cause the T stand for TAP controller, DI and DO stand for data input and data output, so its meaning is tap controller data input (host debugger output )and tap controller data output (host debugger input).
[image]

erhankur, 2022-04-26:
@wuxx Yes, I agree with you. It would be more convenient to swap the TDO/TDI pins. I keep open this issue to follow up.

closed 2022-04-28 by espressif-bot