espressif/esp-usb-bridge #4: LEDs are not initialized

wuxx, 2022-05-07:
Hello, just notice that the LED of serial TX & RX and JTAG are not initialized,  the led cannot blink normally because it is not initialized.
here is my code for reference

```
diff --git a/main/jtag.c b/main/jtag.c
index dab5e48..7cbaf13 100644
--- a/main/jtag.c
+++ b/main/jtag.c
@@ -253,6 +253,15 @@ void jtag_task_resume(void)

 void jtag_task(void *pvParameters)
 {
+
+    gpio_config_t io_conf = {
+        .mode = GPIO_MODE_OUTPUT,
+        .pin_bit_mask = (1ULL << LED_JTAG),
+        .pull_down_en = 0,
+        .pull_up_en = 0,
+    };
+    ESP_ERROR_CHECK(gpio_config(&io_conf));
+
     s_task_handle = xTaskGetCurrentTaskHandle();
     init_jtag_gpio();

diff --git a/main/serial.c b/main/serial.c
index 540e032..0dd7bea 100644
--- a/main/serial.c
+++ b/main/serial.c
@@ -294,6 +294,15 @@ void start_serial_task(void *pvParameters)
         .gpio0_trigger_pin = GPIO_BOOT,
     };

+
+    gpio_config_t io_conf = {
+        .mode = GPIO_MODE_OUTPUT,
+        .pin_bit_mask = (1ULL << LED_TX) | (1ULL << LED_RX),
+        .pull_down_en = 0,
+        .pull_up_en = 0,
+    };
+    ESP_ERROR_CHECK(gpio_config(&io_conf));
+
     if (loader_port_esp32_init(&serial_conf) == ESP_LOADER_SUCCESS) {
         ESP_LOGI(TAG, "UART & GPIO have been initialized");

```

dobairoland, 2022-05-09:
Hi @wuxx The LEDs are initialized in the `init_led_gpios()` function (https://github.com/espressif/esp-usb-bridge/blob/master/main/main.c#L217) executed immediately in the beginning.

Have you tried to set the appropriate GPIO numbers for LED1, LED2 and LED3 in `idf.py menuconfig`?

wuxx, 2022-05-09:
Hi @dobairoland ，yep just see it, but would it be better to modify CONFIG_BRIDGE_GPIO_LED1/CONFIG_BRIDGE_GPIO_LED2/CONFIG_BRIDGE_GPIO_LED3 to LED_TX/LED_RX/LED_JTAG ?
it's my fault that I always hardcode the main/io.h for quick compile ....

dobairoland, 2022-05-09:
Thanks for your feedback but I don't think we will change this right now. There is no official HW for this firmware right now we could follow and one is expected to use it as a template only. Thanks for understanding!

closed 2022-05-09 by dobairoland

wuxx, 2022-05-10:
OK, got it.