espressif/esp-usb-bridge #6: reset target after drag-n-drop program

wuxx, 2022-06-10:
Hi, just notice that after drag-n-drop program, the target chip cannot reset and run automatically. can we optimize it? thanks.

dobairoland, 2022-09-19:
Hi @wuxx. I'm sorry for the late reply. Do you mean that this is just an ESP32-C3 issue? Have you tried another chip or capture the signals on BOOT and RST with a logic analyzer?

The target chip should reset as per https://github.com/espressif/esp-usb-bridge/blob/master/main/msc.c#L424 and https://github.com/espressif/esp-serial-flasher/blob/0b5ca69c99f6607512dc2914351e89bc968dd723/include/esp_loader.h#L156. If it doesn't then it can be an issue with esp-serial-flasher.

It could be also a HW issue. Do you have a pull-up resistor and capacitor on the boot signal for the target MCU? See `M_EN` at
https://github.com/espressif/esp-usb-bridge/blob/master/images/schematics.pdf for an example.

wuxx, 2022-09-21:
hello, @dobairoland got it, I will do more test and check the source code

closed 2023-02-13 by espressif-bot