espressif/esp-usb-bridge #8: target cannot run after drag-n-drop program in ESP32C3

songzhishuo, 2023-01-11:
I have a China ShangHai HeZhou ESP32C3 board, I want to use it try  drag-n-drop program。
 
The target program can be downloaded and run normally through IDF, and the following error occurs after the corresponding uf2 file is downloaded through drag-n-drop. After the development board is newly powered on, it can be operated normally again. I'm wondering why?

Happy Spring Festival！

ESP32C3 Serial print：

`ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x16b4
load:0x3ce00000,len:0x93040
load:0x230027c6,len:0x830ed7ff
Invalid image block, can't boot.
ets_main.c 333 
`

DNedic, 2023-01-13:
Which ```ESP-IDF``` version are you building the project with?

songzhishuo, 2023-01-16:
ESP32C3 and ESP32S2(this device) both in ESP-IDF 4.4

DNedic, 2023-01-17:
Could you please provide ```idf.py monitor``` output from the bridge while flashing?

closed 2023-02-13 by espressif-bot