espressif/esp-usb-bridge #11: Help error program cc2652p via esp32 bridge

newsy5, 2023-05-03:
maybe who knows what the problem might be, I want to download the firmware to ss2652r via esp bridge.

connected it seems to be correct, I use esp32s2
everything connects normally , but the firmware is not upload

config openocd:
adapter driver esp_usb_jtag
espusbjtag vid_pid 0x303a 0x1002
espusbjtag caps_descriptor 0x030A
adapter speed 1000
source [find target/ti_cc26x2.cfg]
set remotetimeout 100
reset_config trst_only
program ptvo.switch-zr02.hex

log openocd:

C:\my\openocd-esp32\bin>openocd
Open On-Chip Debugger v0.12.0-esp32-20230419 (2023-04-18-22:02)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1002
Info : esp_usb_jtag: capabilities descriptor set to 0x30a
Info : esp_usb_jtag: serial (4827E24B37E4)
Info : esp_usb_jtag: Device found. Base speed 480KHz, div range 1 to 1
Info : clock speed 480 kHz
Info : JTAG tap: cc26x2.jrc tap/device found: 0x3bb4102f (mfg: 0x017 (Texas Instruments), part: 0xbb41, ver: 0x3)
Info : JTAG tap: cc26x2.cpu enabled
Info : [cc26x2.cpu] Cortex-M4 r0p1 processor detected
Info : [cc26x2.cpu] target has 6 breakpoints, 4 watchpoints
Info : starting gdb server for cc26x2.cpu on 3333
Info : Listening on port 3333 for gdb connections
Info : JTAG tap: cc26x2.jrc tap/device found: 0x3bb4102f (mfg: 0x017 (Texas Instruments), part: 0xbb41, ver: 0x3)
Info : JTAG tap: cc26x2.cpu enabled
Warn : [cc26x2.cpu] Only resetting the Cortex-M core, use a reset-init event handler to reset any peripherals or configure hardware srst support.
[cc26x2.cpu] halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0xfffffffe msp: 0xfffffffc
** Programming Started **
Info : Padding image section 0 at 0x0000003c with 4036 bytes
Info : Padding image section 1 at 0x000013b8 with 8 bytes
Info : Flash write discontinued at 0x00037b86, next section at 0x00057fa8
Error: libusb_bulk_read error: LIBUSB_ERROR_TIMEOUT
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: esp_usb_jtag: device not found!
Error: esp_usb_jtag: failed to revive USB device!

any ideas what you can try???

dobairoland, 2023-05-03:
Can you share the log from the esp-usb-bridge as well? It would be interesting to see if there are any error reported prior to the error coming from openocd.

Are you able to debug the same device by any other means that the esp-usb-bridge (same device, same openocd)?

newsy5, 2023-05-03:
> Can you share the log from the esp-usb-bridge as well? It would be interesting to see if there are any error reported prior to the error coming from openocd.
> 
> Are you able to debug the same device by any other means that the esp-usb-bridge (same device, same openocd)?

I would love to, but how do I get a log with esp?

> Are you able to debug the same device by any other means that the esp-usb-bridge (same device, same openocd)?
unfortunately I don't have another programmer

dobairoland, 2023-05-03:
> I would love to, but how do I get a log with esp?

The native USB connected to D+, D- is used by the bridge communication (JTAG) in your case. But you can connect to the UART (RxD, TxD) a terminal program. You will see the log there.

dobairoland, 2023-05-03:
> unfortunately I don't have another programmer

I would recommend to try find one. Because you can loose a lot of time try to debug the bridge if the issue is not there.

newsy5, 2023-05-03:
> The native USB connected to D+, D- is used by the bridge communication (JTAG) in your case. But you can connect to the UART (RxD, TxD) a terminal program. You will see the log there.

do you mean that when I connect the esp, I have two devices, one jtag and the second com port? I tried to connect to com port speed 9600 and 115200, but there was nothing on the termen

dobairoland, 2023-05-03:
> do you mean that when I connect the esp,

Two connections are needed, not one.

1. One USB connection through the connector connecting to D+ and D-.
2. Another USB connection to the RxD, TxD signals through an UART-to-USB bridge.

newsy5, 2023-05-03:
> * One USB connection through the connector connecting to D+ and D-.
> * Another USB connection to the RxD, TxD signals through an UART-to-USB bridge.

ok, which pin should I connect to the esp

newsy5, 2023-05-03:
I (21) boot: ESP-IDF v5.0.1 2nd stage bootloader
I (21) boot: compile time 17:24:36
I (21) boot: chip revision: v0.0
I (24) boot.esp32s2: SPI Speed      : 80MHz
I (29) boot.esp32s2: SPI Mode       : DIO
I (33) boot.esp32s2: SPI Flash Size : 2MB
I (38) boot: Enabling RNG early entropy source...
I (44) boot: Partition Table:
I (47) boot: ## Label            Usage          Type ST Offset   Length
I (54) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (62) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (69) boot:  2 factory          factory app      00 00 00010000 00100000
I (77) boot: End of partition table
I (81) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=0acd8h ( 44248) map
I (98) esp_image: segment 1: paddr=0001ad00 vaddr=3ffbf100 size=02each ( 11948) load
I (101) esp_image: segment 2: paddr=0001dbb4 vaddr=40022000 size=02464h (  9316) load
I (109) esp_image: segment 3: paddr=00020020 vaddr=40080020 size=21a90h (137872) map
I (142) esp_image: segment 4: paddr=00041ab8 vaddr=40024464 size=0ac9ch ( 44188) load
I (160) boot: Loaded app from partition at offset 0x10000
I (160) boot: Disabling RNG early entropy source...
I (172) cache: Instruction cache        : size 8KB, 4Ways, cache line size 32Byte
I (172) cpu_start: Pro cpu up.
I (191) cpu_start: Pro cpu start user code
I (191) cpu_start: cpu freq: 160000000 Hz
I (191) cpu_start: Application information:
I (194) cpu_start: Project name:     bridge
I (199) cpu_start: App version:      30b8d93-dirty
I (204) cpu_start: Compile time:     May  3 2023 17:24:40
I (210) cpu_start: ELF file SHA256:  4b3ca6157c003c04...
I (216) cpu_start: ESP-IDF:          v5.0.1
I (221) cpu_start: Min chip rev:     v0.0
I (226) cpu_start: Max chip rev:     v1.99
I (231) cpu_start: Chip rev:         v0.0
I (235) heap_init: Initializing. RAM available for dynamic allocation:
I (243) heap_init: At 3FFC3918 len 000386E8 (225 KiB): DRAM
I (249) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (255) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAM
I (262) spi_flash: detected chip: generic
I (266) spi_flash: flash io: dio
W (270) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (283) esp_core_dump_uart: Init core dump to UART
I (289) cpu_start: Starting scheduler on PRO CPU.
I (294) gpio: GPIO[12]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (294) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (304) gpio: GPIO[14]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (314) bridge_main: LED GPIO init done
I (324) bridge_msc: MSC disk RAM usage: 698 bytes
I (324) uart: queue free spaces: 20
I (334) gpio: GPIO[7]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (344) gpio: GPIO[4]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (354) bridge_serial: UART & GPIO have been initialized
I (354) gpio: GPIO[3]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (364) gpio: GPIO[8]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (374) gpio: GPIO[10]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (384) gpio: GPIO[9]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (394) bridge_jtag: JTAG GPIO init done
I (544) bridge_jtag: Mounted
I (584) bridge_serial: DTR = 0, RTS = 0 -> BOOT = 1, RST = 1
I (24824) bridge_jtag: bRequest: (3) wValue: (0) wIndex: (0)
I (24824) bridge_jtag: bRequest: (0) wValue: (1) wIndex: (0) "!!!! this line appeared during the firmware"

erhankur, 2023-05-04:
@newsy5 Can you run openocd with -d3 option to see the verbose debug logs? Until now, I didn't connect any arm chip to the bridge. I will try to test it soon.  We use esp_usb_jtag protocol for the jtag communication, maybe it requires some adjustment.

newsy5, 2023-05-04:
i test openocd version v0.12.0-esp32-20230419, v0.12.0-esp32-20230313, v0.11.0-esp32-20221026 and v0.12.0 oficial
the problem is always the same

newsy5, 2023-05-11:
please tell me

erhankur, 2023-05-11:
Sorry, couldn't test it yet with the arm board. Can I ask for one more debug? 

Please add the below lines to your config file and send me the log.txt 

`espusbjtag log "log.txt"`
```
adapter driver esp_usb_jtag
espusbjtag vid_pid 0x303a 0x1002
espusbjtag caps_descriptor 0x030A
espusbjtag log "log.txt"
adapter speed 1000
source [find target/ti_cc26x2.cfg]
set remotetimeout 100
reset_config trst_only
program ptvo.switch-zr02.hex
```

newsy5, 2023-05-14:
conole with config openocd.cfg

adapter driver esp_usb_jtag 
espusbjtag vid_pid 0x303a 0x1002
espusbjtag caps_descriptor 0x030A
espusbjtag log "log.txt"
adapter speed 1000
source [find target/ti_cc26x2.cfg]
set remotetimeout 100
reset_config trst_only
program ptvo.switch-zr02.hex

 the problem is in the line `espusbjtag log "log.txt"` everything is as it was without her

adapter driver esp_usb_jtag
espusbjtag vid_pid 0x303a 0x1002
espusbjtag caps_descriptor 0x030A
espusbjtag log "log.txt"
adapter speed 1000
source [find target/ti_cc26x2.cfg]
set remotetimeout 100
reset_config trst_only
program ptvo.switch-zr02.hex

C:\my\openocd-esp32\bin>openocd
Open On-Chip Debugger v0.12.0-esp32-20230419 (2023-04-18-22:02)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1002
Info : esp_usb_jtag: capabilities descriptor set to 0x30a
Error: The 'espusbjtag log' command must be used after 'init'.
openocd.cfg:5: Error:
in procedure 'script'
at file "embedded:startup.tcl", line 28
at file "openocd.cfg", line 5
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Warn : An adapter speed is not selected in the init scripts. OpenOCD will try to run the adapter at very low speed (100 kHz).
Warn : To remove this warnings and achieve reasonable communication speed with the target, set "adapter speed" or "jtag_rclk" in the init scripts.
Info : esp_usb_jtag: serial (4827E24B37E4)
Info : esp_usb_jtag: Device found. Base speed 4800KHz, div range 1 to 1
Info : clock speed 4800 kHz
Warn : There are no enabled taps.  AUTO PROBING MIGHT NOT WORK!!
Error: JTAG scan chain interrogation failed: all zeroes
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: IR capture error at bit 0, saw 0x00 not 0x...3
Warn : Bypassing JTAG setup events due to errors
Warn : gdb services need one or more targets defined

erhankur, 2023-05-14:
Ok. You can run OpenOCD with -c parameter like below. 
```
openocd -f <scripts> -c "init; espusbjtag log "log.txt""
```

newsy5, 2023-05-14:
logs run 
[cc.txt](https://github.com/espressif/esp-usb-bridge/files/11472117/cc.txt)
[log.txt](https://github.com/espressif/esp-usb-bridge/files/11472118/log.txt)

newsy5, 2023-05-25:
have you looked at the log, is there a misunderstanding why it doesn't work?

erhankur, 2023-05-29:
Hi, @newsy5 I am sorry, couldn't figure out why the ARM board disappears during communication.  I need to test it but still don't have a board with the Jtag pins exported.

erhankur, 2023-05-29:
Finally, I found a board with a Texas arm chip. It couldn't be able to complete the first examination process. I am on it. Will let you know when I have a progress.

```
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: serial (7CDFA1E0104C)
Info : esp_usb_jtag: Device found. Base speed 4800KHz, div range 1 to 1
Info : clock speed 4800 kHz
Info : JTAG tap: cc32xx.jrc tap/device found: 0x0b97c02f (mfg: 0x017 (Texas Instruments), part: 0xb97c, ver: 0x0)
Info : JTAG tap: cc32xx.cpu enabled
Error: Invalid ACK (4) in DAP response
Error: JTAG-DP STICKY ERROR
```

newsy5, 2023-05-29:
tnx, i im wait

erhankur, 2023-06-19:
@newsy5 After replicating the issue and conducting further investigation, it appears that we encounter a problem when OpenOCD requests a large amount of flash data for read/write operations, particularly on ARM chips. Considering other priorities at the moment, I will address this issue when I have sufficient time available.