espressif/esp-usb-bridge #13: Working for an ESP32S3?

bamber11999, 2023-05-31:
Hi.

I've been trying to get this project to work on a ESP32-S3-WROOM-1. I purchased two of these devkits and would love to be able to use one as a debug adapter for an esp32 wrover-e (rather than going out and buying a specialized adapter.)  

The S3 has been successfully flashed. However, given the default wiring under the bridge configuration, I haven't been able to flash or debug the esp32.   The M_TDI and M_TDO given in the schematic are switched compared to the config file. 

```
CONFIG_BRIDGE_GPIO_TDI=3
CONFIG_BRIDGE_GPIO_TDO=9
```

Running openocd manually...
```

openocd -f board/esp32-bridge.cfg
Open On-Chip Debugger v0.11.0-esp32-20221026 (2022-10-26-14:48)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1002
Info : esp_usb_jtag: capabilities descriptor set to 0x30a
Info : esp_usb_jtag: target chip id set to 1
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Error: libusb_open() failed with LIBUSB_ERROR_NOT_FOUND
Error: esp_usb_jtag: could not find or open device!
```

[image]

Any help is appreciated.

erhankur, 2023-05-31:
Keep in mind TDI and TDO pins must be configured from the target point of view. 
```
ESP32 - ESP32S3
IO12 - CONFIG_BRIDGE_GPIO_TDI
IO15 - CONFIG_BRIDGE_GPIO_TDO
```

And I suggest updating your OpenOCD version to the last release. We have linked OpenOCD with the latest libusb version for some Windows issues. It could help. 

https://github.com/espressif/openocd-esp32/releases/tag/v0.12.0-esp32-20230419

bamber11999, 2023-06-01:
Openocd had been updated but there was no change.   

Is that JTAG device above meant to be unrecognized?  I used Zadig to install a winusb driver for that device and things got a little better ie openocd stopped crashing.    However,    with the current wiring, there was no way for me to flash or monitor the esp32 let alone start debugging.

erhankur, 2023-06-01:
@bamber11999 let's go one by one. Could you start debugging from the bridge with the preloaded target?

bamber11999, 2023-06-02:
Nope, couldn't.  Same errors as previous before I installed the winusb driver. 

I'm going to close this out.   It's obviously not working out of the box for windows right now and I don't want to spend that much more time on it.  Thanks for your help.

closed 2023-06-02 by bamber11999