espressif/esp-usb-bridge #14: Abstracting the functionality of esp-usb-bridge into components

lijunru-hub, 2023-07-11:
I have recently implemented wireless debugging using the code from this repository. However, maintaining it on both ends can be quite challenging. You can find the code for wireless debugging at this GitHub repository: https://github.com/espressif/esp-dev-kits/tree/master/esp32-s3-usb-bridge/examples/usb_wireless_bridge.

It would be great to have the CDC, JTAG, Serial, and MSC functions abstracted into an API so that we can customize the message queue by calling the interface. Additionally, it would be interesting to explore the possibility of supporting wireless debugging using ESP-NOW.

dobairoland, 2023-07-11:
Thanks @lijunru-hub for the suggestions!

Yecgaa1, 2023-08-19:
@lijunru-hub I noticed your contributions in the [repository,](https://github.com/espressif/esp-dev-kits/tree/master/esp32-s3-usb-bridge/examples/usb_wireless_bridge.) but I noticed that it doesn't seem to implement wireless jtag debugging, and I wonder if you have any plans to implement this feature
Forgive me if this is not the appropriate place to raise this question

lijunru-hub, 2023-10-25:
@Yecgaa1 I'm sorry for the delayed response. Due to speed limitations, I haven't had a chance to test the practicality of wireless JTAG yet.

Yecgaa1, 2023-10-25:
Never mind, could you tell me what the current speed limit is?

> @Yecgaa1 I'm sorry for the delayed response. Due to speed limitations, I haven't had a chance to test the practicality of wireless JTAG yet.

Hedda, 2025-03-04:
> I have recently implemented wireless debugging using the code from this repository. However, maintaining it on both ends can be quite challenging. You can find the code for wireless debugging at [...]

Bump this as Espressif removed code from https://github.com/espressif/esp-dev-kits/tree/master/esp32-s3-usb-bridge/examples/usb_wireless_bridge

@lijunru-hub Any chance you would be willing to refactor your previous work and submit it as a pull request?

https://github.com/lijunru-hub/esp-usb-bridge/commits/master/

You can otherwise still find it in forks like these:

  * https://github.com/koson/esp-dev-kits/tree/master/esp32-s3-usb-bridge/examples/usb_wireless_bridge
  * https://github.com/SuGlider/esp-dev-kits/tree/master/esp32-s3-usb-bridge/examples/usb_wireless_bridge

PS: Off-topic but tip is to also check out this ESP Bridge project from Nabu Casa's Z-Wave JS developer:

* https://github.com/AlCalzone/nc-zwave-esp-bridge/edit/master/README.md