espressif/esp-usb-bridge #15: Issue with programming STM32 via Serial interface

nguyenmanhthao996tn, 2023-07-14:
Hi,

I am trying to program my STM32WLE5 MCU using an ESP32-S3. It's working perfectly for normal read/write serial between my PC and the target MCU. But it's returning error when I try to program/flash new firmware with STM32CubeProgrammer (UART Mode).

This is my simple wiring between the two boards. STM32 DFU mode is triggered manually.
[image]

On my PC, I use STM32CubeProgrammer. When I try to connect via UART mode, the software return as following:
```
  17:13:19 : RTS low
  17:13:19 : DTR low
  17:13:20 : Serial Port COM24 is successfully opened.
  17:13:20 : Port configuration: parity = even, baudrate = 115200, data-bit = 8,                     stop-bit = 1.0, flow-control = off
  17:13:20 : Activating device: OK
  17:13:20 : Board       : --
  17:13:20 : Chip ID: 0x497 
  17:13:20 : BootLoader protocol version: 3.3
  17:13:21 : Response received from device: NACK
  17:13:21 : Error: GETID command not acknowledged!
  17:13:21 : Reemission of GetID command
  17:13:53 : Error: RDP is set to level 1, try using Read UnProtect (-rdu) command in order to remove the Read Out Protection.
  17:13:53 : UPLOADING OPTION BYTES DATA ...
  17:13:53 :   Bank          : 0x00
  17:13:53 :   Address       : 0x1fff7800
  17:13:53 :   Size          : 104 Bytes
  17:13:53 : Timeout error occured while waiting for acknowledgement.
  17:13:53 : Error: Address not acknowledged: 0x1FFF7800
  17:13:53 : Error: Uploading Option Bytes bank: 0 failed
  17:13:53 : Error: Initializing the Option Bytes failed
  17:13:53 : Disconnected from device.
```

I did try it on different PC with both Windows and Linux, it return the same error.
It's also working flawlessly when I try replacing the ESP32-S3 with a CP2104 USB-to-TTL (same wiring setup). So, It's seem not to be a hardware issue.

Further information:
- [esp-usb-bridge](https://github.com/espressif/esp-usb-bridge) commit: 01ed029
- STM32CubeProgrammer version: v2.10.0

DNedic, 2023-07-17:
Hello @nguyenmanhthao996tn , as we don't support flashing other MCUs with the `esp-usb-bridge`, we cannot reproduce your issue. It would help us narrow down the issue if you could provide a logic analyzer or oscilloscope trace.

Also, have you tried powering the `STM32WLE5` MCU separately from the `esp-usb-bridge` board and only connecting ground, to avoid power delivery issues?

nguyenmanhthao996tn, 2023-07-17:
> Hello @nguyenmanhthao996tn , as we don't support flashing other MCUs with the `esp-usb-bridge`, we cannot reproduce your issue. It would help us narrow down the issue if you could provide a logic [...]

Hi @DNedic , thanks a lot for you reply. I don't have a logic analyzer or oscilloscope at home. Let's me try to look for some and provide more information with it.

Regarding to the power of STM32, I did try to seperate them and use the common ground. But, unfortunately, the result stayed the same.

DNedic, 2024-08-28:
Closing this for inactivity.

closed 2024-08-28 by DNedic