espressif/esp-usb-bridge #17: Question: ESP32-C3?

greg80303, 2023-11-09:
Would this USB->UART bridge firmware work on a ESP32 C3 device?

dobairoland, 2023-11-10:
Currently only ESP32-S2 and ESP32-S3 are supported, meaning that we tested only on boards with these chips. 

I don't see why it couldn't work on ESP32-C3. Maybe you will need to do some minor modification in the code.

radimkarnis, 2023-11-10:
AFAIK, only chips with the integrated USB peripheral are supported (ESP32-S2 and ESP32-S3).

ESP32-C3 only has USB-Serial/JTAG, which is implemented in HW and cannot be reprogrammed, therefore cannot support the esp-usb-bridge project.

dobairoland, 2023-11-10:
Thank you @radimkarnis for the correction.

closed 2023-11-10 by dobairoland

igrr, 2023-11-10:
To add, it is possible to use ESP32-C3 as a USB-serial-to-UART adapter, with the limitation that the DTR and RTS signals of the virtual serial port will control ESP32-C3's reset & download mode, and can't be propagated outside the chip.

This design limitation was fixed on ESP32-S3. It is possible to use ESP32-S3's USB_SERIAL_JTAG peripheral to implement a USB to UART JTAG bridge mostly in hardware. The software only needs to forward the data between USB_SERIAL_JTAG and UART interfaces.

However, unlike this project, that solution wouldn't be able to support other USB classes, such as Mass Storage.

greg80303, 2023-11-10:
Thanks for the responses.  I also found [this](https://github.com/aliengreen/esp32_uart_bridge) project, which I've been trying to use.  But so far, I can't get it to work.  I'm going to keep looking at it.

igrr, 2023-11-10:
Looks like that project should indeed work! Make sure that your terminal program is not setting the DTR and RTS signals of the virtual serial port, as that would reset the ESP32-C3.

To debug the issue, I would suggest hooking up another USB-Serial converter to ESP32-C3's UART, and see the log output there while you are trying to connect the USB port to PC.

greg80303, 2023-11-10:
Thanks @igrr !

igrr, 2023-11-15:
@marciogranzotto Can you clarify which _this_ you mean? esp-usb-bridge project or https://github.com/aliengreen/esp32_uart_bridge?

I think the former won't work (again, no USB_OTG peripheral on C6) but the latter should work (C6 has a USB_SERIAL_JTAG).

gcormier, 2025-04-04:
It's too bad it won't work on the C3.

I was able to get the project mentioned (https://github.com/aliengreen/esp32_uart_bridge) to work on the C3, but the performance is very very slow.