espressif/esp-usb-bridge #19: download failed under Windows

wuxx, 2023-12-14:
Hello, in the past, I have been using esp-usb-bridge under Linux without any issues. Just now I found that it cannot be downloaded smoothly on Windows. It seems that there are some bugs in the source code, possibly related to the collaboration with esptool.py. I compiled the latest code, but the same problem persists. The specific situation is as follows:

[image]

The issue seems to occur during the change of the baudrate. If I use the -b 115200 parameter (i.e.,  do not need to change the baudrate), then the programming is successful."

wuxx, 2023-12-14:
Programming with the "-b 115200" parameter is successful.

[image]

wuxx, 2023-12-14:
I just checked the waveform file and source code. The reason is that this function call did not switch the baud rate correctly.

[image]

DNedic, 2023-12-14:
@wuxx Thank you for your report, can you try UF2 flashing with the bridge, and using `esptool` with an USB to serial converter. That would help us better pinpoint the issue, and if indeed this seems to be an `esptool` issue, please submit an issue in the `esptool` repository.

DNedic, 2023-12-14:
Thanks, so there might indeed be an issue in the interaction of `esptool` and `esp-usb-bridge`, I will try reproducing the issue.

wuxx, 2023-12-14:
Okay, I just opened an issue under esptool repo
https://github.com/espressif/esptool/issues/939

DNedic, 2024-01-14:
I have managed to reproduce and diagnose the issue, a fix is on the way.

closed 2024-01-24 by espressif-bot

wuxx, 2024-01-25:
Hi, DNedic, I checked the submitted code, but I still don’t quite understand why it can work normally on Linux but not on Windows,  Can you explain it briefly?

DNedic, 2024-01-25:
> Hi, DNedic, I checked the submitted code, but I still don’t quite understand why it can work normally on Linux but not on Windows, Can you explain it briefly?

Hello @wuxx , the reason is that the windows virtual com port driver keeps sending DTR = 0, RTS = 0 during flashing. Because the `tud_cdc_line_state_cb()` called `serial_set_baudrate()` when DTR and RTS are not both asserted, under an assumption that this will only happen when resetting the device, this causes the baudrate to be erratically changed during flashing. 
 
Now we are checking whether RTS is asserted (translating to reset) and only change the baudrate to default when the device is reset.