espressif/esp-usb-bridge #23: Any way to flash new software without BOOT button?

mickeyl, 2024-02-13:
I have been delievered two esp-usb-bridge hardware devices, one has a broken BOOT button. Is there a way to flash new software to that hardware without the BOOT button being pressable?

dobairoland, 2024-02-13:
Can you please be more specific about the hardware device? I don't think we sell anything right now with esp-usb-bridge but I know about some non-Espressif board manufacturers who sell devices with esp-usb-bridge (usually they have their own issue trackers).

Generally speaking, If the serial RTS & DTR signals are connected to the Reset and Boot pins of an ESP32(-*) device then the device can be flashed with esptool without pressing the Boot button. More about this can be found at https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/boot-mode-selection.html#automatic-bootloader.

mickeyl, 2024-02-13:
https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32s3/esp32-s3-usb-bridge/index.html ­– isn't this an official Espressif device?
[image]

 As far as I understand the schematics, RTS & DTR for the S3 are not available through the pin headers…

dobairoland, 2024-02-13:
Yes, I'm sorry, I forgot about that one. It was done by a completely different team.

If I read [this table](https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32s3/esp32-s3-usb-bridge/user_guide.html#gpio-allocation) well then pin no. 3 is the boot signal for the module. Have you tried to connect it to GND while powering it up (sticking it into the USB slot)?

DNedic, 2024-02-13:
Perhaps it would be best to try [DFU](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/dfu.html) or [esp-usb-jtag](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/jtag-debugging/index.html#upload-application-for-debugging)?

mickeyl, 2024-02-13:
Shortening the boot pin and the ground has no effect on this device.

I might try JTAG.

mickeyl, 2024-02-13:
FWIW, it looks like the latest software/documentation for this device is here: https://github.com/espressif/esp-dev-kits/tree/master/esp32-s3-usb-bridge/examples/usb_wireless_bridge/

dobairoland, 2024-02-19:
I'm closing this because this is not a firmware issue. I can recommend to send back the damaged board if you are not able to use the above recommendations.

closed 2024-02-19 by dobairoland