espressif/esp-usb-bridge #24: Driver error on Windows 11

pieterjanbuntinx, 2024-06-07:
After flashing a ESP32-S3 board with the usb-bridge firmware I get the following error on Windows 11 in device manager:

```
Device settings for USB\VID_303A&PID_1002\F412FA91A04C were not migrated from previous OS installation due to partial or ambiguous device match.

Last Device Instance Id: USB\VID_2572&PID_0012\VMRK2
Class Guid: {36fc9e60-c465-11cf-8056-444553540000}
Location Path: PCIROOT(0)#PCI(0801)#PCI(0003)#USBROOT(0)#USB(2)
Migration Rank: 0xF000FFFFF0000020
Present: false
Status: 0xC0000719
```

A couple of months ago this worked perfectly. 
I have tried using Zadig to force a new driver. None of the driver options worked.

jakub-kocka, 2024-06-11:
Hi @pieterjanbuntinx,

We apologize for the inconvenience. It appears that there is an issue in the firmware related to the `tinyusb` dependency. You can use an older commit to build and flash a working binary: [e42a7f9714c6d9138a5cb76a079d04157a42c657](https://github.com/espressif/esp-usb-bridge/tree/e42a7f9714c6d9138a5cb76a079d04157a42c657).

We are working to resolve this as soon as possible.

masbug, 2024-09-21:
Same issue on Windows 10.
Commit [e42a7f9714c6d9138a5cb76a079d04157a42c657](https://github.com/espressif/esp-usb-bridge/tree/e42a7f9714c6d9138a5cb76a079d04157a42c657) doesn't compile. dcd_esp32sx.c is missing USB_OUT_EP_NUM.
You need to add this to compile:
```
#include "soc/usb_reg.h"
#include "soc/usb_struct.h"
#include "soc/usb_wrap_struct.h"

#ifndef USB_OUT_EP_NUM
#define USB_OUT_EP_NUM ((int) (sizeof(USB0.out_ep_reg) / sizeof(USB0.out_ep_reg[0])))
#endif

#ifndef USB_IN_EP_NUM
#define USB_IN_EP_NUM ((int) (sizeof(USB0.in_ep_reg) / sizeof(USB0.in_ep_reg[0])))
#endif
```
...and for the missing windows driver: https://visualgdb.com/UsbDriverTool/

closed 2024-12-18 by espressif-bot