espressif/esp-usb-bridge #26: 该设备无法启动。 (代码 10)  {操作失败} 请求的操作不成功。

windskyxb, 2024-11-09:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

使用最新代码编译并烧录后它应该能够被系统正常识别并工作

### Actual behavior (suspected bug)

设备管理器提示：该设备无法启动。 (代码 10){操作失败}请求的操作不成功。
[image]
[image]
[image]

### Error logs or terminal output

```shell
该设备无法启动。 (代码 10)

{操作失败}
请求的操作不成功。

由于仅部分匹配或匹配不明确，因此无法迁移设备 USB\VID_303A&PID_1002\F412FAD8F000。

上个设备实例 ID: USB\VID_1D6B&PID_0104\0123456789ABCDEF
类 Guid: {36fc9e60-c465-11cf-8056-444553540000}
位置路径: 
迁移等级: 0xF000FFFF0000F120
存在: false
状态: 0xC0000719
```

### Steps to reproduce the behavior

1. clone此仓库
2. idf.py set-target esp32s3
3. idf.py menuconfig
4. idf.py build
5. idf.py -p COM33 flash

### Project release version

master

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

windows10 22h2

### Shell

CMD

DNedic, 2024-11-12:
Hello @windskyxb , this is a known issue, see https://github.com/espressif/esp-usb-bridge/issues/24.

windskyxb, 2024-11-14:
> Hello @windskyxb , this is a known issue, see #24.

but commit [e42a7f9714c6d9138a5cb76a079d04157a42c657](https://github.com/espressif/esp-usb-bridge/tree/e42a7f9714c6d9138a5cb76a079d04157a42c657) not provided ARM Serial Wire Debug (SWD) device，I would prefer to use the ARM Serial Wire Debug (SWD) device

Dzarda7, 2025-01-03:
Hi @windskyxb this issue should be resolved, you can try SWD and it should work on the latest commit.

windskyxb, 2025-01-03:
> Hi @windskyxb this issue should be resolved, you can try SWD and it should work on the latest commit.

Yes, after pulling the latest code and compiling it, SWD works fine

closed 2025-01-03 by windskyxb