espressif/esp-usb-bridge #27: Build failed(tinyusb error)

wirwolf, 2024-12-07:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Code compiling without error

### Actual behavior (suspected bug)

Code not compiled and idf.py menuconfig not work

### Error logs or terminal output

```shell
..CMake Error at /home/user/esp/esp-idf/tools/cmake/build.cmake:611 (message):
  ERROR: Because project depends on espressif/tinyusb (^0.15.0) which doesn't
  match any versions, version solving failed.

Call Stack (most recent call first):
  /home/user/esp/esp-idf/tools/cmake/project.cmake:735 (idf_build_process)
  CMakeLists.txt:12 (project)

```
```

### Steps to reproduce the behavior

```
git clone ...
idf.py set-target esp32c3
idf.py menuconfig
```

### Project release version

master

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Kbuntu 22.04

### Shell

Bash

### Additional context

ESP-IDF v5.5-dev-811-ga2f420a36e

wirwolf, 2024-12-07:
Also i see in output 
-- Component directory /home/user/esp/esp-usb-bridge/components/tinyusb does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/user/esp/esp-usb-bridge/components/tinyusb does not contain a CMakeLists.txt file. No component will be added

dobairoland, 2024-12-07:
Hi @wirwolf. ESP32-C3 is not supported. Please see https://github.com/espressif/esp-usb-bridge/issues/17 for more information. The main README states that only ESP32-S2 and ESP32-S3 are supported.

As for the build issue, I think it is a local issue. I recommend to check that `git status` is clean and try to run `idf.py reconfigure`.

> -- Component directory /home/user/esp/esp-usb-bridge/components/tinyusb does not contain a CMakeLists.txt file. No component will be added

You can see in the repository that we don't have tinyusb in the components directory. Instead it is placed during the build into the managed_components directory by the component manager. Have you modified the project?

wirwolf, 2024-12-07:
@dobairoland Yes, I can try to add tinyusb, becase if I reset the code to the master branch and run ```idf.py reconfigure``` I see this error
```
CMake Error at /home/user/esp/esp-idf/tools/cmake/build.cmake:611 (message):
  ERROR: Because project depends on espressif/tinyusb (^0.15.0~4) which
  doesn't match any versions, version solving failed.

Call Stack (most recent call first):
  /home/user/esp/esp-idf/tools/cmake/project.cmake:735 (idf_build_process)
  CMakeLists.txt:12 (project)

```

closed 2024-12-09 by dobairoland