espressif/esp-usb-bridge #28: Unable to use `esp-usb-bridge` (Feather ESP32-S2 as Bridge, Sonoff Basic R2 as Target)

5hubh4m, 2024-12-15:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

Hello, I'm trying to use my Feather ESP32-S2 as a USB-UART bridge to flash my Sonoff Basic R2. I setup `esp-idf` v5.3.2 and `esp-usb-bidge` repo and flashed it on my Feather after entering bootloader mode.

```
$> idf.py -p /dev/cu.usbmodem01 flash monitor
Executing action: flash
Running make in directory /Users/shubham/Documents/Code/esp-usb-bridge/build
Executing "make -j 10 flash"...

...

Writing at 0x00008000... (100 %)Error: ESP32-S2FNR2 (revision v0.0) chip was placed into download mode using GPIO0.
esptool.py can not exit the download mode over USB. To run the app, reset the chip manually.
To suppress this note, set --after option to 'no_reset'.

Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.0 seconds (effective 553.8 kbit/s)...
Hash of data verified.

Leaving...
CMake Error at run_serial_tool.cmake:66 (message):
  
  /Users/shubham/.espressif/python_env/idf5.3_py3.9_env/bin/python;;/Users/shubham/Documents/Code/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32s2
  failed.

make[3]: *** [CMakeFiles/flash] Error 1
make[2]: *** [CMakeFiles/flash.dir/all] Error 2
make[1]: *** [CMakeFiles/flash.dir/rule] Error 2
make: *** [flash] Error 2
make failed with exit code 2, output of the command is in the /Users/shubham/Documents/Code/esp-usb-bridge/build/log/idf_py_stderr_output_81309 and /Users/shubham/Documents/Code/esp-usb-bridge/build/log/idf_py_stdout_output_81309
```

Since it cannot automatically reset the Feather, I manually reset. After the reset the COM port disappears, and the Feather just has the power LED on. I can see the USB bridge in System Information but I cannot find a port to monitor the Feather's serial debugger.

[image]

```
$> ls /dev/cu*
/dev/cu.Bluetooth-Incoming-Port  /dev/cu.URT0
```

I also cannot see a COM port for when I connect my Sonoff to the Feather (pin 5 to Sonoff's Rx and pin 6 to Sonoff's Tx --- the default assignment from the schematics), I've also tried to connect them the other way around and it's the same.

What am I missing?

Dzarda7, 2024-12-17:
Hi @5hubh4m, thanks for reporting this. Regarding the error message, I will take a look at it, but it should not be the problem for you. Now regarding the port disappearance. It should be enough to change `espressif/tinyusb:` to `"^0.17.0~1"` instead of  `"^0.15.0~4"` in the main/idf_component.yml file and rebuild the project. Please let me know if it works, fix is on the way.

Dzarda7, 2024-12-17:
The error message is intentional as esptool is unable to reset the target when flashed using usb otg, so it prints error, but flashing should proceed, you only need to reset the target by yourself. This will also be fixed in new esptool.

closed 2024-12-18 by espressif-bot