espressif/esp-usb-bridge #29: FT2232 MPSSE emulation wished

emard, 2025-01-04:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Described the feature in detail and justified the reason for the request.
- [X] Provided specific use cases and examples.

### Feature description

Most closed source programming JTAG tools assume FT2232/232H/4232
MPSSE as a "driverless" de-facto standard for JTAGing. Some ebay dongles
claiming altera-xilinx-lattice compatibility actually have STM32 emulating
FT2232.

So my wish for esp32s3/c3 would be to emulate FT2232 mpsse
and become "driverless" wide "standard" JTAG.

One attempt at it was done but left unfinished

https://github.com/mithro/fx2-ftdi-emulation

### Use cases

"driverless" compatibility with existing closed source tools.
No need to write new driver for JTAG for open source tools.

Dzarda7, 2025-01-06:
Hi @emard, thanks for the suggestion. We will probably not implement this unfortunately as it is a bit out of scope of this repo. Also if I understand correctly, the FT2232 and the dongles can be used. Are you able to present some usecase other than replacing these already made chips?

emard, 2025-01-06:
yes it's significantly out of range just I wanted to share some thoughs.
FT2232 is great and fast, especially widely supported almost every JTAG
software supports FT2232 well and in more or less limited functionality
other chips

To work, FT2232 needs companion chips like power supply,  eeprom
and other small parts so all this takes PCB space. When PCB wants to be
small there's just no space for FT2232 but maybe we'd like to keep
JTAG software working as if we had ft2232, this is the reason.

when I was designing a board there was no space for FT2232 so
FT231X was placed. 7 years every today now and
then some issues come related to hacky usage of FT231X as
JTAG and ESP32 programmer, some on windows, some on apple,
only linux works mostly ok

As board already has ESP32, possible upgrade could be to place ESP32S3
and emulate some subset of FT2232 just to get jtag and serial working,
Idea is to emulate just a subset, not everything 2232 really has.

Dzarda7, 2025-01-07:
I will close this issue unfortunately as we will not implement it in this repo, but will keep it in mind, if it (or something similar) ever comes out again as a request and we might create the project, but I don't want to get your hopes up. Anyway thanks for the suggestions.

Dzarda7, 2025-01-12:
I took a more closer look at the FT2232 as I took this issue as an inspiration and I think it might make sense to add features like SPI and I2C as a feature to this project. This probably will not help you much, but it might help someone, so thanks for this.

emard, 2025-01-13:
of course, great to have is always spi (with multiple CS for
few devices), i2c, jtag and 2 serial ports.
You have to design it optimally because there is limited
number of endpoints available so instead of reinventing
it+s best to choose some chip which already has kernel
and application drivers and all this features already built in