espressif/esp-usb-bridge #30: Micropython version of esp-usb-bridge wished

emard, 2025-01-04:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Described the feature in detail and justified the reason for the request.
- [X] Provided specific use cases and examples.

### Feature description

Stock micropython binaries (esp32s3) are compiled with
tinyusb support (not hardware cdc-jtag) that allow creating
custom usb devices from micropython.

If we want to integrate micropython and usb-jtag-bridge
one option is to compile micropython with hardware cdc-jtag,
other is to have usb-jtag-bridge written in micropython.

### Use cases

Intelligent usb-jtag-bridge that is not only JTAG but also
convert protocol converter/adapter with wifi bluetooth.
Often hardware debugging needs additional hardware
next to jtag, with this programmable jtag we can have
many-in-one device.

Dzarda7, 2025-01-07:
Hi @emard, thanks for the suggestion, but isn't this more like request for micropython team? But if I understand correctly, you would like to have something like esp-usb-bridge, but implemented in micropython?

emard, 2025-01-07:
yes yes, something like esp_usb_bridge.py in micropython
would be cool to have. But it's not a hurry, I have still
various ideas how could I program onboard fpga with usb
esp32s3 which runs micropython. You can also close this
issue and/or place it in some future todo

Dzarda7, 2025-01-12:
I will close this, but I think you might try to take it to micropython repo and suggest it to them with the link to this repo for inspiration. I do not think it makes much sense for us to maintain two repos with the same purpose, sorry for that.

closed 2025-01-12 by Dzarda7