espressif/esp-usb-bridge #32: S3 vs S2 devkits

gcormier, 2025-04-03:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

Hello,

I have two official Espressif devkits. S2-MINI and S3-MINI.

On S3 mini, tool is working great. I have GPIO5 + 6 as RX/TX, and I have a wire connecting those 2 together, so a serial connection to COMx on Windows will provide terminal echo as expected.

On an S2 mini, with GPIO5+6 connected together, I am not getting any output. I am able to connect to the COM port without any issues, but no output is observed.

Am I doing something wrong?

Dzarda7, 2025-04-04:
Hi @gcormier, can you tell me which devkit of S2 do you have? I believe we do not sell some S2-MINI, nor S3-MINI devkits. Anyway, as long as there is no added component to the pin, it should work. If your devkit does have USB-UART converter, you can read the boot message, if not, you can try to connect it to the RX and TX pins of the S2 to see if the initialization ends up correctly, it should be on pin 43 and 44. I would also double check the pin definition again. If none of this is an issue, I will try to investigate it next week as I will have access to S2. Thanks for your report.

gcormier, 2025-04-04:
Sorry, you are correct, the S3 is not a Mini, but the S2 is.

ESP32-S3-DevKitC-1 v1.1
ESP32-S2-DevKitM-1 v1.0

I can do more investigation this weekend as well.

Dzarda7, 2025-04-04:
It makes sense it does not work then. This project uses native USB of the chips. The S3 devkit you have has 2 USB ports, one is connected to the native USB of the chip, the second one is connected through USB-UART converter (cp2102) to the UART of the chip. The S2 devkit you have has only one USB port and it is the UART one, which is not utilized for purpose of this project (just some logging and debugging). If you want to use the S2 devkit, you need to connect GPIO19 and GPIO20 directly to USB through some exposed USB cable. These pins are labeled as USB_D+ and USB_D- as you can see [here](https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32s2/esp32-s2-devkitm-1/user_guide.html#pin-layout). Or use different devkit.

gcormier, 2025-04-04:
Sorry, I misread schematic. I thought it was using native USB, but I see now that two resistors are unpopulated.

[image]

Therefore, I can move R1/R4 to R2/R3 to bypass CP2102.

Dzarda7, 2025-04-04:
Oh, yes, nice, did not know there are pads ready 😄. It should work then after this modification.

gcormier, 2025-04-04:
I will report back after the modification.

Will this project work on the esp32-c3?

gcormier, 2025-04-04:
Confirmed working on S2 after swapping the resistors!

I will ask about C3 on a different discussion to keep things organized.

closed 2025-04-04 by gcormier

dobairoland, 2025-04-07:
> Will this project work on the esp32-c3?

Please see https://github.com/espressif/esp-usb-bridge/issues/17