espressif/esp-usb-bridge #37: ESP-Prog 2 Using ESP32S3

Alaa-Aim, 2025-08-22:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

I tried to import the the files from https://github.com/espressif/esp-usb-bridge , to Espressif-IDE and failed, I used CLI to compile and flash the firmware to ESP32S3, and I'm trying to use it to debug and ESP32, when I chose Debug, below errors shows and nothing work 

'Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
This GDB was configured as "--host=x86_64-w64-mingw32 --target=xtensa-esp-elf".
Type "show configuration" for configuration details.
<https://www.gnu.org/software/gdb/bugs/>.
http://www.gnu.org/software/gdb/documentation/>.

Info : accepting 'gdb' connection on tcp/3333
Error: Target not examined yet
Please refer to the troubleshooting guide below to identify the problem.
https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
Error executing event gdb-attach on target esp32.cpu0:
embedded:startup.tcl:1561: Error: 
in procedure 'program_esp' called at file "F:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\share\openocd\scripts/target/esp_common.cfg", line 562
in procedure 'program_error' called at file "F:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\share\openocd\scripts/target/esp_common.cfg", line 395
at file "embedded:startup.tcl", line 1561
Warn : No symbols for FreeRTOS!
Error: Target not halted
Please refer to the troubleshooting guide below to identify the problem.
https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
Error: auto_probe failed
Please refer to the troubleshooting guide below to identify the problem.
https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Please refer to the troubleshooting guide below to identify the problem.
https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
Error: attempted 'gdb' connection rejected
Please refer to the troubleshooting guide below to identify the problem.
https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
shutdown command invoked'

and when I chose run, below errors shows and nothing flashed 

'Open On-Chip Debugger v0.12.0-esp32-20240318 (2024-03-18-18:26)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1002
Info : esp_usb_jtag: capabilities descriptor set to 0x30a
Info : esp_usb_jtag: target chip id set to 1
Flashing "F:/Espressif/WorkPlace/blink/build/bootloader/bootloader.bin" at 0x1000
** program_esp input args <0x1000 verify> **
Info : esp_usb_jtag: serial (7CDFA1F49150)
Info : esp_usb_jtag: Device found. Base speed 4800KHz, div range 1 to 1
Info : clock speed 4800 kHz
Error: JTAG scan chain interrogation failed: all zeroes
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: esp32.cpu0: IR capture error; saw 0x00 not 0x01
Warn : Bypassing JTAG setup events due to errors
Warn : [esp32.cpu0] Unexpected OCD_ID = 00000000
Error: [esp32.cpu0] Examination failed
Warn : target esp32.cpu0 examination failed
Warn : [esp32.cpu1] Unexpected OCD_ID = 00000000
Error: [esp32.cpu1] Examination failed
Warn : target esp32.cpu1 examination failed
Info : starting gdb server for esp32.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Error: JTAG scan chain interrogation failed: all zeroes
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: esp32.cpu0: IR capture error; saw 0x00 not 0x01
Warn : Bypassing JTAG setup events due to errors
Warn : [esp32.cpu0] Unexpected OCD_ID = 00000000
Warn : [esp32.cpu1] Unexpected OCD_ID = 00000000
Info : [esp32.cpu0] requesting target halt and executing a soft reset
Error: Target not examined yet
Info : [esp32.cpu1] requesting target halt and executing a soft reset
Error: Target not examined yet
Warn : [esp32.cpu0] Unexpected OCD_ID = 00000000
** Flashing Failed **
-1
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections'

Please advise.

Dzarda7, 2025-08-22:
Hello @Alaa-Aim, can you please invoke ESP-IDF environment in your terminal and try to run `openocd -f board/esp32-bridge.cfg`?

Alaa-Aim, 2025-08-22:
> Hello [@Alaa-Aim](https://github.com/Alaa-Aim), can you please invoke ESP-IDF environment in your terminal and try to run `openocd -f board/esp32-bridge.cfg`?

Hello @Dzarda7 , I did, and this is what i got 
"

PS F:\Espressif\frameworks\esp-idf-v5.3.1> openocd -f board/esp32-bridge.cfg
Open On-Chip Debugger v0.12.0-esp32-20240318 (2024-03-18-18:26)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1002
Info : esp_usb_jtag: capabilities descriptor set to 0x30a
Info : esp_usb_jtag: target chip id set to 1
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: esp_usb_jtag: could not find or open device!
F:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\share\openocd\scripts/target/esp_common.cfg:9: Error:
at file "F:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\share\openocd\scripts/target/esp_common.cfg", line 9
PS F:\Espressif\frameworks\esp-idf-v5.3.1> "

Dzarda7, 2025-08-22:
@Alaa-Aim is it possible that you either have selected wrong pinout when flashing esp-usb-bridge or wrongly connected the target to esp32s3?

Alaa-Aim, 2025-08-22:
> You have wrong pinout, the one you are setting using menuconfig refers to the one on ESP32-S3 and you need to connect it to the corresponding pins on ESP32 (MTDI, MTMS, MTCK, MTDO), as you can see the blue ones in the [pin description](https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32/_images/esp32_devkitC_v4_pinlayout.png).

So you mean I have to use the default configurations ? as shown below ? 

[image]

and use the table below for connection
 
F:\Espressif\WorkPlace\esp-usb-bridge-1.1.0

from the official schamatic for Esp-prog 2 as shown below 

[image]

am i right ?

Alaa-Aim, 2025-08-22:
> I cannot unfortunately, I have never worked with Espressif-IDE, I tried it now, but it did not really work properly on my machine. If you want, you can try VScode plugin.

@Dzarda7 , many thanks, I'll try use VSCode

Alaa-Aim, 2025-08-22:
> I cannot unfortunately, I have never worked with Espressif-IDE, I tried it now, but it did not really work properly on my machine. If you want, you can try VScode plugin.

@Dzarda7 No luck with VSCode 😢😥, I couldn't get it work

Dzarda7, 2025-08-25:
@Alaa-Aim happy to hear that!